OrangeFox86/DreamPicoPort #10: Add USB Mass Storage Device for VMU Memory

Tails86, 2022-10-01:


hpx1984, 2022-11-18:
Hi Sir
how to use this function?

Tails86, 2022-11-18:
It's an open item. I haven't done any development for it yet, but I
envision setting up a mass storage device with maybe directory structure of
all files found. Or maybe just the raw data files of each VMU which could
then be imported into an emulator.

Tails86, 2022-11-20:
I'm diving into it now - probably another week or two, depending on how much free time I find. FAT16 drive is setup with placeholders for the raw memory files. It's very likely that these files can be dumped right to/from redream vmu bin files (haven't checked that yet, but the file size of redream files is exactly the size of memory available in a VMU).
[image]
[image]

closed 2022-11-25 by Tails86

Tails86, 2022-11-25:
I tweaked things a bit so that files only show up when the associated memory unit is attached. I confirmed that the file is compatible with redream. Files at the moment are read only.

[image]
[image]