OrangeFox86/DreamPicoPort #12: DC controller no functional when test in Window10 and XP's game controller test App

hpx1984, 2022-11-14:
Dear Sir

i used p2040 board and Main.uf2, when this board to PC(connect 1set DC controller on the board) ,the game controller test app can saw 4items "4P"
but my DC controller no functional on game controller test APP
please help me for fix it if you did know how to fix the problem
[image]

Tails86, 2022-11-14:
Thank you for your interest in this project!

I assume you are using this release, right? https://github.com/OrangeFox86/DreamcastControllerUsbPico/releases/tag/v0.1

The LED on the board should be on once connected to USB and off when any button is pressed on the controller or the analog stick is moved. That should be a quick test to make sure basic controller communication is working.

I have been mainly using Windows 11 to develop this and https://gamepad-tester.com/ to quickly test things out. You should see P1, P2, P3, P4 there and pressing buttons should register. 

I have a feeling that Windows is going to have trouble labeling the 4 controllers right in some circumstances (possibly with older versions) which is why it looks like P4 is labeled for all on your system. I may put in a fix in the future to only enumerate connected controllers so that it will at least look fine when a single controller is connected. I can get you a custom single player variant of the uf2 in the meantime if it seems like that is the issue here. How many controllers are you trying to wire up at the moment?

hpx1984, 2022-11-14:
Hi Sir,
thanks your replied any supported ,
yes， i used this FW on my board
the LED always on when i pressed any button
i want connect 4 controllers but at the first i just connected one for test , if it is working fine then i will go on for connect more controller 
i used a controller extension cable for soldered on PI Board and then connected my 1st controller on it 

i got this but still no action on the page when i pressed button
[image]

Tails86, 2022-11-14:
Looks like that page enumerates the controller properly. So far so good!

It sounds like you're not getting communication with the controller.

Check your wiring. You made me realize that I forgot to document the pinout at the port. Refer to the image here: http://www.raphnet-tech.com/products/dreamcast_usb_circuit/dc_connector_pinout.png
1: SDCKA
2: +5V
3: GND
4: GND
5: SDCKB

That should correlate to the schematic here:
https://raw.githubusercontent.com/OrangeFox86/DreamcastControllerUsbPico/main/images/schematic.png

hpx1984, 2022-11-15:
hi Sir
double checked the connection it is correct 
now i cut the extension cable for short the cable on connection but the system still not working,
any i tried two boards also got same problem 
maybe i need to dis-assemble the controller for make sure it is the sega official controller 
i had 8sets DC controller on hand but all can't working on this system , but its is working on DC
i will try to find where the problem ,if you have any comment please also let my know for try   

[image]
[image]

Tails86, 2022-11-15:
I see where the confusion is. The schematic I used called out physical pin numbers, not the GPIO pin numbers. The top image is what my schematic was drawn for. 
14 -> GP10
15 -> GP11
16 -> GP12
17 -> GP13
24 -> GP18
25 -> GP19
26 -> GP20
27 -> GP21
[image]

Looks like the bottom one you're using lists out the actual GPIO pin numbers, so 10-13 and 18-21 would be the right pins on that one.

I'll update my schematic to make that clearer as well.

hpx1984, 2022-11-15:
Hi Sir

Great
now the system can working，many thanks your great help
 
now seems perfect if can fix the 4-P4 issue 

[image]

Tails86, 2022-11-15:
I've seen this happen before when using generic HID gamepad descriptors, and it seems to be a long-known bug with Windows DirectX/drivers from forum posts I have gone through. It still happens in Windows 11 now that I've actually looked. What's worse is that Windows doesn't even seem to sort them in the proper order. I'm assuming that the web interface is grabbing raw data instead of going through DirectX which is why things show up properly at gamepad-tester.com.

I _think_ my only options here would be to either use a different USB descriptor for the device like xinput or spin up my own driver. Using xinput seems to be what most controller manufacturers are doing these days in order to avoid writing drivers. I started playing with xinput, but then I got distracted with other things. It will likely be a few more months before this gets fixed.

[This project](https://github.com/fluffymadness/tinyusb-xinput) will likely be very helpful.

hpx1984, 2022-11-18:
many thanks your quickly support ，Sir
the function is good that is first important, so this just a very smaller problem

but this project really a great project for let peoples enjoy DC game on PC like DC console   
and this project has great potential as one day it may grow to a family project for support other game console ,like PSXControllerUsbPico, NGCControllerUsbPico, N64ControllerUsbPico,PCEControllerUsbPico,3DOControllerUsbPico...........
hope can see in future

Tails86, 2022-11-27:
Opened up https://github.com/OrangeFox86/DreamcastControllerUsbPico/issues/18 to fix that bug, and I'm closing this thread out

closed 2022-11-27 by Tails86