OrangeFox86/DreamPicoPort #18: Windows doesn't parse players 1 through 4 labels properly

Tails86, 2022-11-27:
It seems to be a long-known bug with Windows DirectX/drivers from forum posts I have gone through. It still happens in Windows 11 now that I've actually looked. What's worse is that Windows doesn't even seem to sort them in the proper order. I'm assuming that the web interface is grabbing raw data instead of going through DirectX which is why things show up properly at gamepad-tester.com.

I think my only options here would be to either use a different USB descriptor for the device like xinput or spin up my own driver. Using xinput seems to be what most controller manufacturers are doing these days in order to avoid writing drivers.

[This project](https://github.com/fluffymadness/tinyusb-xinput) will likely be very helpful.

multigamesystem, 2023-11-01:
I'm also seeing similar behaviour in RetroArch via Linux.  Which is breaking functionality please make it so only one controller is recognized when one controller is plugged in.  The same issue is happening that when one controller is plugged in it is automatically set to P4.  I've only wired one controller input up as according to your readme.

Tails86, 2025-01-25:
I'm moving forward with just supporting compatibility through the serial interface to whatever emulator wants to interface with this (flycast support is in dev at the moment, but only for 1p mode). I don't think there is anything that can be done with how Windows handles things.