OrangeFox86/DreamPicoPort #108: Add a suggestion for the 2-bit Bus Transceiver chip (isolation circuitry)

ErykDarnowski, 2023-09-10:
For a novice in eletronics (like me) I'd be really helpful to see an example of the needed 2-bit Bus Transceiver. I've found a suggestion in this issue's comment: https://github.com/OrangeFox86/DreamcastControllerUsbPico/issues/70#issuecomment-1445623241 but it would be great if it were added to the `README.md` for someone in the future to see right away!

P.S.
I'm utterly amazed with this project! I've recently got interested in emulation and (among others) got myself a Dreamcast controller. I'd love to use the **VMU**, **Jump Pack** and perhaps even the **Microphone ** with an emulator, and this project (as far as I've seen) is my only hope, so huge props and congratulations on your work, hope to see more soon 😄

Ohh, and can't wait to build the hardware!

Tails86, 2023-09-10:
That is fair. I got a bit sidetracked over the last few months and didn't come back to flesh that documentation out. I have been using 74LVC2T45DC which is made by Nexperia and TI. I tied both Vcc inputs to 3.3 V, the DIR feeds to pin 5, and the signals feed to pins 2 and 3. 

Where I bought them from:
https://www.digikey.com/en/products/detail/nexperia-usa-inc/74LVC2T45DC-125/2530595
https://www.digikey.com/en/products/detail/texas-instruments/SN74LVC2T45DCUR/657833

This is the one bit that makes this project less accessible since these chips only come in surface mount, 0.5 mm pitch SOP8. I found these little adapter boards which at least help with prototyping. It's still a giant pain to work with such small leads though.
[image]

I'll likely be able to find the time next month to start working on this project again. I hit a wall with trying to get the force feedback through USB working on Windows. It may be easiest to make the device look like an XBox controller, which seems to be what companies like 8BitDo do to get compatibility with Windows.

ErykDarnowski, 2023-09-11:
Sounds great!

Have you looked in to using **DirectInput** / **XInput**? That's the 2 PC input solutions my `SNES -> USB` Mayflash adapter provides (I'm pretty sure their other adapters rely on those too).

## Resources

- [image]
- [What's the difference between Direct-Input and XInput](https://www.reddit.com/r/8bitdo/comments/q8zjxf/comment/hgts8q0/?utm_source=share&utm_medium=web2x&context=3)
- [Getting Started With XInput in Windows applications](https://learn.microsoft.com/en-us/windows/win32/xinput/getting-started-with-xinput)
- [Introduction to DirectInput](https://learn.microsoft.com/en-us/previous-versions/windows/desktop/ee418273(v=vs.85))

Tails86, 2023-09-11:
Yep! That's what I was talking about - XInput is what XBox controllers use. The main reasons I have been avoiding that is because it will force me to use specific VID/PID, and I would no longer be able to enumerate 4 controllers at once since I didn't see a way to add a hub in the current USB API.

ErykDarnowski, 2023-09-11:
Hmmm... **XInput** should support up to 4 controllers, but there's **DirectInput** too (which doesn't have such limitation)!

## Resources

- <https://gaming.stackexchange.com/a/358003>

Tails86, 2023-09-11:
What I was describing are limitations of the USB API at the device level. XInput requires certain things to be on certain endpoints and such, so I can't specify multiple controllers on the same device (like I'm currently doing with HID gamepad). In order to tell the host that there are multiple XInput controllers, you need to have it come up as 4 separate devices - this requires telling the host that my device is a USB hub, and then enumerate 4 separate USB devices on that hub. The USB API doesn't seem to allow for this kind of config yet (at least from the last time I checked a few months back, and maybe there is something I overlooked). It would be technically possible though... I am just not prepared to put in that effort unless it was already available in tinyusb 😆

Tails86, 2023-09-11:
This is as I suspected. No current support, and no support is planned:
https://github.com/hathach/tinyusb/discussions/1904

ErykDarnowski, 2023-09-11:
Ohhh! My bad, that's not good news! Also, thank you for the detailed explanation 😥

closed 2023-11-19 by Tails86