OrangeFox86/DreamPicoPort #110: cant compile it im not good at this lol

secretman28, 2023-11-02:
i wana change pin for remote toeasly and not use the bottom pin  wana change pin 
// The start pin of the two-pin bus for each player
#define P1_BUS_START_PIN 10
#define P2_BUS_START_PIN 12
#define P3_BUS_START_PIN 2
#define P4_BUS_START_PIN 4
can some cone compile the thing and get me the u2f that i can flash it 

or tell me what im doing wrong i can compile anything helse my vscode is set right

Tails86, 2023-11-02:
This is untested. Remove the .txt extension after downloading.
[host.uf2.txt](https://github.com/OrangeFox86/DreamcastControllerUsbPico/files/13233661/host.uf2.txt)

secretman28, 2023-11-02:
tanks! my freind rape windows to make it work in power shell vscode just dont want i dont know what im doing wrong, i can build any other pico project

closed 2023-11-19 by Tails86