OrangeFox86/DreamPicoPort #112: L trigger gets stuck held when playing Soul Calibur with Redream + Demul

multigamesystem, 2024-01-04:
When pressing the L trigger button while using Redream and playing Soul Calibur the L trigger gets stuck held until you reset the game or close the emulator.  Here is a video recreating the bug.

[https://youtu.be/HHU5TVLXMjk?si=yc71App9An33cQ1G](url)

ErykDarnowski, 2024-01-04:
Hey I think I encountered something similar.

In my case I tried to map some inputs in an emulator but every time I tried to remap something, one of the triggers would **force** itself in as it acted like it was constantly depressed (a state different than idle).

I took a look at the inputs in the Control Panel (like you did) and got the same weird behavior like the one in your YouTube video. Later I compared how the triggers acted to an **Xbox 360 controller** and noticed that they didn't start from the left (`0%`), but from the middle (`50%`) of the bar instead. That's when I decided to implement a quick fix and did [this](https://github.com/OrangeFox86/DreamcastControllerUsbPico/commit/48d4b3424466667c109ab892d5eb057045f1ab61).

multigamesystem, 2024-01-07:
The problem seems to not be a an issue with Retroarch oddly or at least I have noticed it being an issue when testing for the behavior with Soul Calibur.  Hopefully someone figures out a fix for this.

ErykDarnowski, 2024-01-07:
Ohhh...
Well, sorry to say I have no idea how Retroarch behaves with it (I just run my emulators on Windows).
I may test it when I have some free time however.

ErykDarnowski, 2024-01-08:
Ahh! My bad, I thought Retroarch was a Linux distro, but now I've learned that it's just a program.
But, if that's the case, than they issue should persist 🤔.

Tails86, 2024-01-22:
Possibly an interaction with USB HID controller somewhere in the OS or Retroarch?

To give a bit more context, [here](https://github.com/OrangeFox86/DreamcastControllerUsbPico/blob/42cbdd3c1b96ec0701aa757932964f9e78c66e10/src/hal/Usb/Client/Hid/UsbGamepadDreamcastControllerObserver.cpp#L29) is where the value gets loaded into the USB interface. Then [here](https://github.com/OrangeFox86/DreamcastControllerUsbPico/blob/42cbdd3c1b96ec0701aa757932964f9e78c66e10/src/hal/Usb/Client/Hid/UsbGamepad.cpp#L83) is where the value is saved to the descriptor. This is using tiny USB's definition for HID controller [here](https://github.com/hathach/tinyusb/blob/938cae818f5f53887ad12e34189e6746dbce4889/src/class/hid/hid_device.h#L329) which limits the value between 0x81 and 0x7f. The host is supposed to recognize the first value as a 2's compliment negative value, so -127 to 127 should be what it recognizes the range as.

Tails86, 2024-02-03:
Showing the raw value on the calibration screens in the Windows gamepad tool provide the expected values. I tried Soul Calibur in Redream and wasn't able to reproduce.

I tried setting logical min/max to 0 to 255, but that caused more compatibility issues. I instead tried updating the minimum value from -127 to -128 and pushed that - it better correlates to Dreamcast mapping anyway. Maybe that will help? The Windows calibration tool is still jumpy, so I get the feeling there is some bug in Windows code.

multigamesystem, 2024-02-05:
> Showing the raw value on the calibration screens in the Windows gamepad tool provide the expected values. I tried Soul Calibur in Redream and wasn't able to reproduce.
>
> I tried setting logical [...]

Could you link me to the updated .uf2 to test?

Tails86, 2024-02-05:
I published a new release with separate 4p and 1p variants
https://github.com/OrangeFox86/DreamcastControllerUsbPico/releases/tag/v0.13

Tails86, 2025-01-25:
Version 0.14 changes the range of the triggers (z values) to [0,127] and sends value in range [0,127]. This circumvents the compatibility issues I was talking about before while still keeping this in the positive range. I'm hoping that it will be interpreted as "positive" value by anything using it, but it seems the version of Windows and its drivers matters. Maybe now that 0 is considered "released" instead of -127 or -128, it may fix this issue?

multigamesystem, 2025-01-26:
> Version 0.14 changes the range of the triggers (z values) to [0,127] and sends value in range [0,127]. This circumvents the compatibility issues I was talking about before while still keeping this [...]

Thanks for the update and Ill test under Linux (Batocera).

Tails86, 2025-02-10:
I uninstalled and reinstalled the driver several times, and then it suddenly started happening to me too for both triggers in flycast. They will suddenly be interpreted as 50-100% pressed. I want to say this is a weird SDL interaction because the values being reported are the same as they've always been. Deleting [this line](https://github.com/flyinghead/flycast/blob/master/core/sdl/sdl_gamepad.h#L296) makes it start to function like it did before. I'll dig into this more this week.