OrangeFox86/DreamPicoPort #126: Dreamcast or PC ? What is this ?

ShdwDncR, 2025-01-21:
Is this mod to be used on PC's or is this for Dreamcast mod ? Why isn't this project clear about it ?

Tails86, 2025-01-21:
See [README.md](https://github.com/OrangeFox86/DreamcastControllerUsbPico/blob/main/README.md)

> Maple Bus emulation platform for interfacing to a Dreamcast controller port (client mode) or from a Dreamcast peripheral (host mode)

> Host mode allows you to connect up to 4 Dreamcast controllers to a PC over USB.

> Client mode emulates a single controller for use with a Dreamcast. This was added in as an extra feature for this project mainly to demonstrate that the MapleBus library may be used in either direction.

closed 2025-01-21 by Tails86

ShdwDncR, 2025-01-23:
Thx for the info, I was hoping to was a mod to have a Dreamcast controller used on a PC.