OrangeFox86/DreamPicoPort #127: Maple Bus occasionally becomes unresponsive using Flycast

kosekmi, 2025-01-22:
Flycast PR: https://github.com/flyinghead/flycast/pull/1810
Originating PR: https://github.com/OrangeFox86/DreamcastControllerUsbPico/pull/124#issuecomment-2605298783:

I noticed that sometimes the Maple Bus seems to be unresponsive to new messages. I.e., VMU LCD won't update, and Rumble stops working despite Maple Messages being send. The easy fix is re-connecting USB and everything works fine again. This observation is not new to the latest builds, I noticed this from the start.

I now found a way to reproduce this: On Sonic Adventure 2, select 1P Play -> Story -> Hero -> The Beginning (Or Stage Select -> City Escape). Skip cutscene, and start the game. From this point on, Maple Bus is unresponsive. To trigger LCD Update for example, quit SA2, and start any other Game - LCD won't update.
Interestingly, it keeps being responsive for serial commands like "h" which triggers the help. Then again, Maple messages being send outside Flycast (e.g., send default VMU screen via some serial terminal software) also won't trigger an LCD update.

So I figured that the handling within the microcontroller may have an issue processing Maple messages. While I am still lacking a proper debug setup for the Pico, maybe this is something you could try to reproduce and take a look?

Edit:
Just checked on Win 11 23H2 where this isn't reproducible as stated above. On MacOS 15.2 it is.

kosekmi, 2025-01-22:
Response from @Tails86 in  https://github.com/OrangeFox86/DreamcastControllerUsbPico/pull/124#issuecomment-2605298783

yeah, ok. It sounds like something going on with the peripheral handling somewhere then. I'll try to reproduce. If you are able to, could you maybe log the serial messages being sent/received and their timestamps?

I only have Linux/Windows environments I can play with over here.

kosekmi, 2025-01-22:
I generated a Maple Message Log capturing the issue: [sa2_debug.log](https://github.com/user-attachments/files/18510631/sa2_debug.log)

Setup: MacOS 15.2, Sonic Adventure 2. VSCode addon "Serial Monitor" sniffs on the serial port. Debug build of host-1p which enables serial port mirroring.

1. Load Save State with Level of Rogue: VMU LCD changes to Rogue
2. Running around the Level, getting hit 2 times which triggers Rumble
3. Pause Game, select Exit. Title Screen appears: VMU LCD changes to SA2 Logo
4. VMU LCD SA2 Logo is send every ~10-30ms in the title screen
5. Shortly after, the VSCode addon does not log any more messages
6. Maple Bus is unresponsive: No VMU LCD Update or Rumble, and no message mirroring
7. Cross check: Stop Flycast, move to the VSCode addon and send any Maple Message (e.g., "X 0E 02 00 02 00 00 01 00 10 F0 3B 05" for Rumble). Same behaviour as 6.
8. Reconnect USB - everything is working again

kosekmi, 2025-01-22:
The "coincidence" with the high frequency of messages got me thinking, and I can now reproduce the issue with a simple setup.

```
std::string values[] = {"X 0E 02 00 02 00 00 01 00 10 F0 3B 05\r\n", "X 0E 02 00 02 00 00 01 00 10 00 0F 00\r\n"};
int index = 0;

while (true) {
    asio::async_write(serial_handler, asio::buffer(values[index]), asio::transfer_exactly(values[index].size()), [ &serial_handler](const asio::error_code& error, size_t bytes_transferred)
    {
        if (error) {
            serial_handler.cancel();
        }
    });
    DEBUG_LOG(INPUT, "%s", values[index].c_str());
    index = (index + 1) % 2;
    std::this_thread::sleep_for(std::chrono::milliseconds(1));
}
```

The Code sends Rumble [On/Off] in quick succession. When sleeping 10ms between messages everything is fine, when using 1ms the Maple Bus becomes unresponsive after ~3s. So it seems it comes down to message frequency, probably in conjunction with their size.

kosekmi, 2025-01-22:
Addition to the setup: When using larger messages, e.g., sending VMU LCD Messages like below (oscillating between 2 images for visual comparison), the issue is reproducible on a 30ms timer. So message frequency in conjunction with their size seems to be the issue.

Note: we need to use asynchronous writes in Flycast to prevent emulation limiting, so this might also factor in.

`std::string values[] = {"X 0C 01 00 32 00 00 00 04 00 00 00 00 68 00 3F C0 02 01 FC 01 C0 3C 0C 03 FE 06 00 07 98 07 6F 08 00 71 F0 0E FF 10 00 0C 60 1F FF 60 00 02 30 3F 6D C0 00 01 18 36 FF 83 01 F0 14 7F FF BD 06 0E 10 7F 6D CE 38 07 A0 F6 FF 8B C8 07 C0 FF C3 0F 0C 05 40 7F 1F 1F 0E 05 60 7E 67 FD 0B 87 3F C7 8F EB 0D FF F8 F1 33 55 0A BF FF 3D 46 AB 0D 55 63 E1 0F 55 0A AA A0 7B 3A AA 05 55 60 21 E3 56 06 AA A0 3C 82 AA 05 55 60 27 07 56 02 AA A0 21 06 AA 03 55 40 20 07 54 01 AA C0 20 0D AC 00 D5 40 60 1C F8 00 6A 80 70 14 10 00 3F 00 50 22 00 00 18 00 48 22 00 00 0E 00 88 41 00 00 00 00 84 41 00 00 00 01 04 80 80 00 00 01 02\r\n", "X 0C 01 00 32 00 00 00 04 00 00 00 00 00 00 01 FF 80 00 00 00 1F FF F0 00 00 00 7F FF FC 00 00 41 FF FF FE 00 00 63 FF FF E3 00 00 77 FF FF F8 80 00 7F FF FF FE 00 FF FF FF FF FF FF FF E2 A3 6D 1B 15 80 1B 23 65 D1 11 80 12 2B 61 15 55 80 1A AA 29 D5 51 FE 12 2A 2D 15 9B E0 3F FF FF FF FF C0 30 E2 31 C3 C1 80 70 62 11 81 81 80 F0 22 11 00 81 8F FE 22 01 08 8F 80 3E 22 01 18 81 80 3C 22 21 10 F1 C0 30 62 21 00 81 E0 30 E2 31 81 81 FF F1 E2 31 C3 83 FF FF BF FF 7F FF 01 E0 00 00 0C 00 00 B0 00 00 18 00 00 DE 00 00 30 00 00 63 E0 00 60 00 00 30 00 01 C0 00 00 1C 00 07 00 00 00 07 00 7C 00 00 00 01 FF C0 00 00\r\n"};`

Tails86, 2025-01-22:
Thanks for the detailed report! I'll find some time this week to dig through it.

kosekmi, 2025-01-23:
Minimal test which forces the issue using the same `async_write()` handlers as in Flycast: [test.zip](https://github.com/user-attachments/files/18517744/test.zip)

Run on Linux (Debian/Ubuntu):
1. `sudo apt install libboost-all-dev`
2. Update serial port in Line 37
3. `g++ -o test test.cpp -lboost_system -lpthread`
4. Execute 

By default the `Rumble` test case is used, change to `VMU LCD` in lines 8-16

Tails86, 2025-01-24:
I was able to replicate in a very specific way.

I tried creating a Python equivalent, [dctest.py.txt](https://github.com/user-attachments/files/18529176/dctest.py.txt), but I couldn't replicate this issue through Python on either Windows or Linux.

I tried compiling your sample code on Windows (using Conan package manager), but that didn't replicate it there.

My native Linux machine is Ubuntu18 which contained versions of boost and cmake that were too for me to use the code directly or use the most recent Conan release, so I gave up there. (I'll update that later 😝 )

I was able to attach the controller to WSL on my windows machine by opening powershell in Administrator and executing:
```
usbipd list
```
I noted the bus ID of the device labeled `USB Input Device, USB Mass Storage Device, USB Serial Dev...` then executed:
```
usbipd wsl attach --busid <ID>
```
It showed up under my WSL instance as `/dev/ttyACM0`. I compiled your sample code, and I think I finally replicated this!

What's odd for me is if I kill execution and restart it, the controller starts responding again for a while. This didn't sound like your experience though. I assumed that you need to remove and reattach to get things working again. Can you confirm?

Tails86, 2025-01-24:
I'm sort of wondering if this has anything to do with this issue?
https://www.boost.org/doc/libs/1_79_0/doc/html/boost_asio/reference/async_write/overload1.html
> The program must ensure that the stream performs no other write operations (such as async_write, the stream's async_write_some function, or any other composed operations that perform writes) until this operation completes.

kosekmi, 2025-01-24:
> The program must ensure that the stream performs no other write operations (such as async_write, the stream's async_write_some function, or any other composed operations that perform writes) until this operation completes.

> This seems to be the more appropriate way of doing things: [test.cpp.txt](https://github.com/user-attachments/files/18530733/test.cpp.txt)

> But of course, this is less asynchronous as desired. The service needs to be reset every time it runs out of work. I remember dealing with asio in the past, and I'm just now realizing THAT is why I had issues 🤦. I have had luck with adding a dummy boost::asio::io_service::work to an io_service just to make sure it never runs out of work. Then you can just run io_service.run() in another thread. Call io_service.stop() when it's time to exit. You'll still need to ensure you're not calling async_write before the previous write has completed. There isn't anything wrong with calling async_write from the callback that gets triggered though (I've done this without issue)

Thanks much, that seemed to be the issue! Closing this issue here, will update Flycast later

closed 2025-01-24 by kosekmi