OrangeFox86/DreamPicoPort #171: Host: Power supply issue when using Raspberry Pi Pico Zero

rpf16rj, 2025-02-19:
Hi, i using a Raspberry Pi Pico Zero and when i connect to PC and connect a Dreamcast Controller, the USB is disconnect and Windows shows the warn:

"Unknow USB device need more power of than provided capacity."

I verified the wire connection of DC port and is correct. 

[image]

[image]

[image]

Tails86, 2025-02-19:
I see that the device is popping up which rules out mis-wiring, and I can see from your provided pictured that it is wired up right. I have received another report recently that getting a powered USB hub may be necessary to properly power the controller with the VMU and purupuru attached. @nexus382, can you comment on that one?

rpf16rj, 2025-02-19:
But in my case, only detect when don't connect DC controller on rp2040. When i connect DC controller, he disconnect and show warn of power supply, without VMU and Rumble connected. I will test with a powered hub and back.

Thanks.

Tails86, 2025-02-19:
Ah! In that case, I suggest you check your wiring. I seem to have similar color wires, and on mine yellow is +5v, green + gray are ground. I think what is happening is a direct short from +5 V to ground occurs when you plug the controller in.

Tails86, 2025-02-19:
You should be referring to the left side of this image for the socket side connecting to the Pico.
[image]

I have a feeling that you may have mirrored things because my SDCKA and SDCKB colors to gp10 and gp11 are reversed from yours on my cable. I'll update some of this documentation to be a bit more clear in the future.

rpf16rj, 2025-02-19:
Oh my gosh, you are right,  i was inverted 5v pin with GND, i need take more attenption. Thanks for advise me.
Thank you for help and congratulations for this project.

Now is working:

[image]

[image]

[image]

rpf16rj, 2025-02-19:
Sure, can keep open.

This one VMU with blacklight is a little dark, i has another that blacklight is better.

nexus382, 2025-02-21:
im actually trying to do this for the vmus i have, and for a few friends, which guide/item list todid u go by to do yours if you do not mind me asking.

rpf16rj, 2025-02-21:
> im actually trying to do this for the vmus i have, and for a few friends, which guide/item list todid u go by to do yours if you do not mind me asking.

Sure, basically, I took a broken LCD screen from an old phone and removed the leds and screen difusors papers. I cut it for the screen size of VMU and made a wire connection to the leds from the 5v pin from VMU, with a 200omhs resistor to reduce the light power.

I follow this guide:
https://www.youtube.com/watch?v=7OAy7e_PmBo

rpf16rj, 2025-02-26:
> yes! so I must use a powered hub with my pico2, and when using my rumble AND my vmu. so to make sure were working fine, Just use the VMU, ensure proper functionality... because i can rock a VMU no issue... and if im not mistaken, I can do rumble with no VMU, but sucking enough power to rock that motor AND the vmu, to much with out a powered pico.
> 
> also.. we using the resistors? try removing them if you are. otherwise, just go get your self a cheapass powered usbhub, (a hub that plugs in to a power outlet)

Hi, I'm back again. I noticed that in some games that use Purupack in an insane way, like Ferrari 350, when the Purupack begins to rumble, the joypad restarts (the VMU screen shows the QRcode again).
I tested with connected directly with a USB and a powered hub USB.

Tails86, 2025-02-27:
Thanks, I'll try to replicate that this weekend

Tails86, 2025-03-02:
I tested a few different things. I couldn't replicate this through my docking station or directly connected to my laptop. This does sound like what @nexus382 was running into. You may need to try a different powered hub or maybe even power it with a beefier external power source.

I measured current draw when vibration was going crazy, and I peaked at 190 mA. I can't really get instantaneous readings though, so there may be a larger spike that I'm not picking up.

I did run into trigger issues due to interactions between SDL2 and DInput on Windows. I applied a workaround on Feb 16. Try the latest and let me know if you still have trigger issues: https://github.com/flyinghead/flycast/actions/runs/13564228724

Tails86, 2025-03-03:
You could also try putting a capacitor between 5v and ground on the pico

Tails86, 2025-03-03:
Yeah, the more I think about it, the more I feel like a capacitor would be necessary to run some of these jump packs. If they included a capacitor inside the jump pack, they may have dried up by now. Something like 100 uf at 15V (or higher) would probably be more than enough to handle any dips. I'll do some more testing to see if I can replicate this on other systems/jump packs and see if this fixes the issue.

rpf16rj, 2025-03-04:
> I tested a few different things. I couldn't replicate this through my docking station or directly connected to my laptop. This does sound like what [@nexus382](https://github.com/nexus382) was [...]

Hi @Tails86, thanks for your answer. I tried this version of flycast, same issue occurs.

I will try another power supply for my USB hub. 

I already tried putting a 100uF capacitor on 5v and gnd of pico, but I think that didn't resolve this issue, so I'll try again.

Tails86, 2025-03-04:
Interesting! You might want to try artifacts here: https://github.com/OrangeFox86/DreamPicoPort/actions/runs/13658788047/artifacts/2690290692

I'm working on improving many things within this branch. A couple of changes which may help your situation:
1. The startup vibration was removed which may be causing reboot loops in some situations when input power is restricted
2. The disconnect timeout value was extended from about 30 ms up to about 160 ms

rpf16rj, 2025-03-05:
> Interesting! You might want to try artifacts here: https://github.com/OrangeFox86/DreamPicoPort/actions/runs/13658788047/artifacts/2690290692
>
> I'm working on improving many things within this [...]

I just tried this build and wit capacitor 100uf, same issue. 
I dont' noticeed reconnect, but the right trigger stop work and some times, the analogic stick is like reset the center position, for eg: When this ocurres, if i was pressing the right trigger, the full pressed position is 0% presset, when release trigger, nothing happing. 
And if I was movin left stick to right for eg and relese left stick, he keep moving to right.

Tails86, 2025-03-05:
I'd like to get more data points on this issue. Do you mind seeing if this still happens when no VMU is attached and just Purupuru pack? If you still have the issue with that, can you try with just the VMU?

rpf16rj, 2025-03-05:
> I'd like to get more data points on this issue. Do you mind seeing if this still happens when no VMU is attached and just Purupuru pack? If you still have the issue with that, can you try with just the VMU?

With just VMU attached, this issue doesn't occur. With only Purupack attached, the problem occurs. 
I have a USB power meter like this https://myvolts.com/imgs/prd/656/0/S-HC-0516_4.jpg and noticed that delivery power is around 4.7v with my USB hub powered. So as my USB power input used an old power cable, I soldered a USB-C female connector directly to use my best USB cable and power supply. Now I have 5.01v and 117mAh showing on USB meter screen, and the issue ocurrs again. 
I don't think that is a low power issue. Maybe the rumble rotation does energy back again to Pico? I already see some circuits using a diode to previne somethink with this.

Tails86, 2025-03-05:
On the regular pico board, the difference between VBUS and VSYS is that VSYS is 5V after the on-board diode. I think the zero doesn't have a diode. It wouldn't hurt to try throwing one in there if you have a diode laying around.

nexus382, 2025-03-06:
can I ask and share what my test have shown?

When using a volt meter i seem to be around 3.01-3.4 V ocming thru for my controller and pico 2. htisis what is running when i test the grounds to the 5v line. I am very sure this is an AMPERAGE not VOLTAGE issue.

When im not on a pwoered hub, i use crazy taxi as crazy dashes and limit breaks and i can ake my controllre go from High to AX vibration on my Tremmor pack which is also a MU so that is oe of the most power hungry. If i ru the vmu AND the rumble, when the rumble starts up or when its going nusts this can happen. If i unplug the VMU and leave the Ruble/MU it rarly happens. When im plugged in to my good powered hub, none of these happen.

this tells me a few thigs.

the restart that happens super fast ad pops up the mass storage window like when you plug the vmu/cntrolelr in, will occure, when i hit my amperage limit. and this is what you seem to be having.

other then this, if you have tried a different powreed hub, and stil get this, its somethng speciic with the zero, maye the diod above thats being mentioned? would be nice if you had multiple controlers and VMUS to try to pull more power with out a motor t otest your diod idea.

rpf16rj, 2025-03-06:
Did test with a diode in5839 between 5v pin of pico zero and 5v pin of controller, the issue ocurres again. I did a new test with comon rp2040 pico and back results.

Tails86, 2025-03-07:
I don't really have much of an idea as to what could be happening here. Just summarizing what I know:
- When a motor first turns on, there is normally a large power draw to get it running (static voltage, high current as nexus382 describes).
    - You've tried adding a capacitor to help power the motor during this period
- When a motor turns off, there might possibly be a voltage kick back as the motor spins down (inductance). I would assume that a nicely built jump pack would account for this in its electronics, but who knows?
    - You've tried adding a diode to ensure these spikes are not fed back to the pico or USB port.
    - (The capacitor should help a bit with this too - to absorb and smooth out these spikes)
- What you are describing sounds likes the USB interface is completely disconnecting which means either the USB port is reacting to a large power draw and shutting off the port or the pico itself is momentarily shutting off
    - You've tried a completely separate voltage source without success

At this point, I'm wondering if it's your purupuru pack. Does this function properly in a real Dreamcast?

rpf16rj, 2025-03-09:
My Purupack works normally, but it is true that it isn't original, but on Dreamcast real hardware, I don't have any problem.

rpf16rj, 2025-03-09:
Today i will test with Raspberry Pico instead zero version.

Tails86, 2025-03-30:
This documentation is hopefully a little less confusing: https://github.com/OrangeFox86/DreamPicoPort/wiki/Installation-Guide#for-host-mode

I did realize that I never called out that top ground wire (what is labeled as 6 in the link above). I wonder if not having that connected would cause issues.

ShdwDncR, 2025-12-11:
Got it working without the resistors. Works great under my MisterFPGA, but under RetroArch dose not work properly. Mapping the controller under RetroArch is buggy. I can't map the D-pad under RetroArch, the D-pad just dose not want to get mapped, but I was never interested in RetroArch, as long as it works under MisTerFPGA I'm happy. I don't see any donation buttons here! Where can I tip you ? Also I would recommend you continue the project and make more firmware for the rest of the consoles. I think RasPi is the way. Thanks for making this project.

Tails86, 2025-12-11:
I'm getting the impression that dpad HAT button codes are not as universal as I had hoped. Someone else had issues on Mac outside of Flycast usage. I may think about adding an option to the [settings](https://orangefox86.github.io/DreamPicoPort/webusb/) to use buttons for the dpad instead of that code.

Thanks for the kind words 🙂 I'm not really interested in accepting donations at the moment. Giving Sega equipment a second chance at life is my goal here 😁 

WELCO
METOT
HENEX
TLEVEL