OrangeFox86/DreamPicoPort #182: Please help me Pico

Hunk4TH, 2025-06-08:
Hey guys,

I bought a Pi Pico and a Dreamcast extension cable and kinda f'd up by pulling the wires from the connector before labling them.  Is there a way I can find the right wires with maybe a meter tester or?

Tails86, 2025-06-08:
You can go grab a multimeter with a continuity tester [like this one](https://a.co/d/2mMLuD2). You could even program the pico to be a continuity tester using its LED.

Hunk4TH, 2025-06-08:
> You can go grab a multimeter with a continuity tester [like this one](https://a.co/d/2mMLuD2). You could even program the pico to be a continuity tester using its LED.

Awesome! I have one of those! So using this should point me in the right direction on where to solder the wires? I'm kicking myself for not lsbing them :/ sorry if this is a dumb question. I've only used a multimeter a few times.

Tails86, 2025-06-08:
If your multimeter has this symbol: [image]
it should beep whenever you touch two contacts with the same potential. With that, you can poke a contact of the connector with one end then touch each wire until you hear a beep. If you don't have a continuity tester on your multimeter, you can just set it to resistance (ohm) (Ω) and monitor the display for 0 instead of listening for a beep.

nexus382, 2025-06-10:
ok 1
Did u install the CORRECT softtware?

2.. you DID verify PROPERLY the conncetions? bc it sounds like maybe a wire issue

3. Can u send a picture, and giev me more explination on what setup you are using? any resistors?

4. try with no rumble/vmu, see if flycast sees the device in th controllers page with nothing plugged in  to it

Hunk4TH, 2025-06-13:
Since I don't use the forth port on my Dreamcast, I've been tempted to remove it for this project but would rather not. I'm kinda desperate 😂

Tails86, 2025-06-13:
In case this helps, this is the color coding of cables I buy from Amazon 
[image]
[image]

Hunk4TH, 2025-06-14:
Sweet! I might buy those then. Do you have a link? That image will definitely help! I'm sorry if I was being a pain.

Tails86, 2025-06-14:
That would be this one
https://a.co/d/0Ti3X9a

Hunk4TH, 2025-06-15:
I got it wired up maybe I'm still missing something? The vmu lights up and beeps but that's it nothing else. 

[image]
[image]

Hunk4TH, 2025-06-15:
I was wondering if the wire being long would be an issue myself. I tried swapping with no luck. I'm gonna try again tomorrow. Maybe I should use some wick to remove all the wires and try again fresh? I figured having wires on the Pico would be more convenient instead of soldering directly from the extension cable.

Yes, the Pico led does light up and in the USB game controller settings it even picks it up as a "DreamPicoPort" same with the latest Flycast.

nexus382, 2025-06-16:
ok we need you to do this.,

login to discord and send me a friedn request
Nexus222 

we need to tlk back and forth easier then this and we will solve it in a minute or 2.

closed 2025-07-12 by Tails86